pimylifeup/MFRC522-python #1: My RC522 don't work

fabioce, 2018-04-14:
Hello,
I'm trying this tutorial
https://pimylifeup.com/raspberry-pi-rfid-rc522/
Where is linked this code.
I verified many times the connections, but my sensor is always with red led active and Write.py or Read.py don't write or read the rfid card.
Is possible that my sensor is broke?
There are other tests to evaluate my sensor?
thanks

HumbleBee14, 2018-04-14:
Well it has RED Light only, so no issue with that. I think it blinks when it reads/writes.
I had also used the exact same code & pin-out connections & it worked smoothly.
Just recheck the connections/pins with the reference pin diagram again and Try to take out pins & insert them again with Pi.
It it still doesn't work, then probably there'd be some issue in your reader.

Hey, are you getting any error while running/reading on Pi?
Make sure, your are using GPIO.cleanup() at correct locations! (At the end of program execution)

HumbleBee14, 2018-04-14:
If you have already checked the pins, then only option you have got to check the RFID reader is by using it in some other microcontroller like ARDUINO.
Just check it there with a simple code if it's working with Arduino or not.

fabioce, 2018-04-14:
I don't have an Arduino, only a Raspberry PI 3. I tryed many times to 
reconnect all PINs, but nothing to do.

Your documentation isn't good for me, because I have only a Raspberry.

HumbleBee14, 2018-04-14:
Well, then wait for the Author of this page & this simplified library.
I had used this library as it is as mentioned on that page and it worked on the first run itself.
Can you mention the error you are getting?

fabioce, 2018-04-15:
My error is that the script don't stop when I move RFIDcards near 
sensor, so the RFID sensor don't work.

I think the sensor don't work properly.

tlmyasirs, 2018-05-02:
i have the same problem. but it works fine with arduino. then what ?

closed 2019-03-27 by death-droid