pimylifeup/MFRC522-python #4: Code cannot be installed from Pip

proffalken, 2019-03-19:
We're looking to use this across multiple devices, and installing from Pip would be incredibly useful to us.

It does mean re-arranging some of the code, but the advantage is that we can then build this into our Raspberry Pi config management tooling and deploy it wherever we need to.

death-droid, 2019-03-20:
Pushed changes, and uploaded to pypi.

Should now be able to install using sudo pip install mfrc522

Let me know if there are any issues with this as its my first time dealing with uploading packages to pypi

proffalken, 2019-03-20:
Looks good to me - thanks!

```
[mmw@rincewind ~]$ mkvirtualenv test
Using base prefix '/usr'
New python executable in /home/mmw/.virtualenvs/test/bin/python
Installing setuptools, pip, wheel...done.
virtualenvwrapper.user_scripts creating /home/mmw/.virtualenvs/test/bin/predeactivate
virtualenvwrapper.user_scripts creating /home/mmw/.virtualenvs/test/bin/postdeactivate
virtualenvwrapper.user_scripts creating /home/mmw/.virtualenvs/test/bin/preactivate
virtualenvwrapper.user_scripts creating /home/mmw/.virtualenvs/test/bin/postactivate
virtualenvwrapper.user_scripts creating /home/mmw/.virtualenvs/test/bin/get_env_details
(test) [mmw@rincewind ~]$ pip install mfrc522
Collecting mfrc522
  Downloading https://files.pythonhosted.org/packages/9a/2b/4dc07711dbfb6c89e0e6ea6bb4ba07e293ac43d77df528c852db3b47bfbe/mfrc522-0.0.3-py3-none-any.whl
Collecting RPi.GPIO (from mfrc522)
  Downloading https://files.pythonhosted.org/packages/af/2f/407b6e4cc8a0bdf434825a160bba1807991886b63cce16a5f1a6e1f24cdf/RPi.GPIO-0.6.5.tar.gz
Collecting spidev (from mfrc522)
  Downloading https://files.pythonhosted.org/packages/fb/14/4c2e1640f0cb04862c76d9d76ed7c945b0f67876e503ac02f7f675fe86a0/spidev-3.4.tar.gz
Building wheels for collected packages: RPi.GPIO, spidev
  Building wheel for RPi.GPIO (setup.py) ... done
  Stored in directory: /home/mmw/.cache/pip/wheels/6d/fc/be/f0bd634b0033499def11c5c8b63731c3de443eed482dbe4c71
  Building wheel for spidev (setup.py) ... done
  Stored in directory: /home/mmw/.cache/pip/wheels/10/d6/98/ba1f1999099e3e7adb3a0140af8c9287ff2b067ff005485674
Successfully built RPi.GPIO spidev
Installing collected packages: RPi.GPIO, spidev, mfrc522
Successfully installed RPi.GPIO-0.6.5 mfrc522-0.0.3 spidev-3.4
(test) [mmw@rincewind ~]$ python
Python 3.7.2 (default, Jan 10 2019, 23:51:51) 
[GCC 8.2.1 20181127] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from mfrc522 import SimpleMFRC522
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mmw/.virtualenvs/test/lib/python3.7/site-packages/mfrc522/__init__.py", line 1, in <module>
    from .MFRC522 import MFRC522
  File "/home/mmw/.virtualenvs/test/lib/python3.7/site-packages/mfrc522/MFRC522.py", line 23, in <module>
    import RPi.GPIO as GPIO
  File "/home/mmw/.virtualenvs/test/lib/python3.7/site-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!
```
(Error at the end is because I'm running it on my laptop!)

death-droid, 2019-03-20:
Glad to hear, I have gone and updated the Pi My Life Up tutorial to utilize the pypi package as well.

PR #5 (2019-03-20): Package the software for installation via pip/Pypi
This code is designed to fix #4
closed 2019-03-20 by death-droid