pimylifeup/MFRC522-python #7: No module named MFRC522

matze1708, 2019-03-26:
Hello,

i try to install the MFRC522 Modull.

I have following the instruction from your Link https://pimylifeup.com/raspberry-pi-rfid-rc522/

but i become this error message: Traceback (most recent call last):
  File "read_mfrc522.py", line 6, in <module>
    reader = SimpleMFRC522()
  File "/usr/local/lib/python3.5/dist-packages/mfrc522-0.0.3-py3.5.egg/mfrc522/SimpleMFRC522.py", line 14, in __init__
AttributeError: module 'MFRC522' has no attribute 'MFRC522'

death-droid, 2019-03-26:
Pushed an update that should fix this issue.

Please try updating the MFRC522 package and let us know whether the issue still happens.
`pip install mfrc522 --upgrade`

matze1708, 2019-03-26:
sudo pip install mfrc522 --upgrade
Requirement already up-to-date: mfrc522 in /usr/local/lib/python2.7/dist-packages
Collecting spidev>=3.2 (from mfrc522)
  Downloading https://files.pythonhosted.org/packages/fb/14/4c2e1640f0cb04862c76d9d76ed7c945b0f67876e503ac02f7f675fe86a0/spidev-3.4.tar.gz
Building wheels for collected packages: spidev
  Running setup.py bdist_wheel for spidev ... done
  Stored in directory: /root/.cache/pip/wheels/10/d6/98/ba1f1999099e3e7adb3a0140af8c9287ff2b067ff005485674
Successfully built spidev
Installing collected packages: spidev
  Found existing installation: spidev 3.3
    Not uninstalling spidev at /usr/lib/python2.7/dist-packages, outside environment /usr
Successfully installed spidev-3.4
pi@raspberrypi:~ $ cd Schiessbuch/
pi@raspberrypi:~/Schiessbuch $ python3 read_mfrc522.py 
Traceback (most recent call last):
  File "read_mfrc522.py", line 6, in <module>
    reader = SimpleMFRC522()
  File "/usr/local/lib/python3.5/dist-packages/mfrc522-0.0.3-py3.5.egg/mfrc522/SimpleMFRC522.py", line 14, in __init__
AttributeError: module 'MFRC522' has no attribute 'MFRC522'

death-droid, 2019-03-26:
Hi @matze1708

As you are using python3 and not python 2.7 try.

`sudo python3 -m pip install mfrc522 --upgrade`

You are still running the older version of the package.

matze1708, 2019-03-26:
Hello,

thanks for your help.

I tried this in diffrent Folders. In Folder pi-RFID there is the MFRC522 Folder from this github repostiory.

<sudo python3 -m pip install mfrc522 --upgrade
Collecting mfrc522
  Downloading https://files.pythonhosted.org/packages/fe/0a/db2a498521738d2a5333f196c2e041ade1685bbf36328c87ec1401415c21/mfrc522-0.0.4-py3-none-any.whl
Collecting spidev (from mfrc522)
  Downloading https://www.piwheels.org/simple/spidev/spidev-3.4-cp35-cp35m-linux_armv7l.whl
Collecting RPi.GPIO (from mfrc522)
  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.6.5-cp35-cp35m-linux_armv7l.whl (66kB)
    100% |████████████████████████████████| 71kB 922kB/s 
Installing collected packages: spidev, RPi.GPIO, mfrc522
  Found existing installation: spidev 3.3
    Not uninstalling spidev at /usr/lib/python3/dist-packages, outside environment /usr
  Found existing installation: RPi.GPIO 0.6.3
    Not uninstalling rpi.gpio at /usr/lib/python3/dist-packages, outside environment /usr
  Found existing installation: mfrc522 0.0.3
    Uninstalling mfrc522-0.0.3:
      Successfully uninstalled mfrc522-0.0.3
Successfully installed RPi.GPIO-0.6.5 mfrc522-0.0.4 spidev-3.4
pi@raspberrypi:~ $ cd Schiessbuch/
pi@raspberrypi:~/Schiessbuch $ python3 read_mfrc522.py 
^CTraceback (most recent call last):
  File "read_mfrc522.py", line 9, in <module>
    id, text = reader.read()
  File "/usr/local/lib/python3.5/dist-packages/mfrc522/SimpleMFRC522.py", line 19, in read
    id, text = self.read_no_block()
  File "/usr/local/lib/python3.5/dist-packages/mfrc522/SimpleMFRC522.py", line 38, in read_no_block
    (status, TagType) = self.READER.MFRC522_Request(self.READER.PICC_REQIDL)
  File "/usr/local/lib/python3.5/dist-packages/mfrc522/MFRC522.py", line 249, in MFRC522_Request
    (status, backData, backBits) = self.MFRC522_ToCard(self.PCD_TRANSCEIVE, TagType)
  File "/usr/local/lib/python3.5/dist-packages/mfrc522/MFRC522.py", line 207, in MFRC522_ToCard
    n = self.Read_MFRC522(self.CommIrqReg)
  File "/usr/local/lib/python3.5/dist-packages/mfrc522/MFRC522.py", line 152, in Read_MFRC522
    val = self.spi.xfer2([((addr << 1) & 0x7E) | 0x80, 0])
KeyboardInterrupt
pi@raspberrypi:~/Schiessbuch $ cd ..
pi@raspberrypi:~ $ cd pi-RFID/
pi@raspberrypi:~/pi-RFID $ python3 read_mfrc522.py 
python3: can't open file 'read_mfrc522.py': [Errno 2] No such file or directory
pi@raspberrypi:~/pi-RFID $ cd MFRC522-python/
pi@raspberrypi:~/pi-RFID/MFRC522-python $ python3 read_mfrc522.py 
Traceback (most recent call last):
  File "read_mfrc522.py", line 4, in <module>
    from mfrc522 import SimpleMFRC522
  File "/home/pi/pi-RFID/MFRC522-python/mfrc522/__init__.py", line 2, in <module>
    from .SimpleMFRC522 import SimpleMFRC522
  File "/home/pi/pi-RFID/MFRC522-python/mfrc522/SimpleMFRC522.py", line 3, in <module>
    import MFRC522
ImportError: No module named 'MFRC522>

matze1708, 2019-03-26:
I try to open the example READ file.
I still get this error message

python3 read_mfrc522.py 
Traceback (most recent call last):
  File "read_mfrc522.py", line 4, in <module>
    from mfrc522 import SimpleMFRC522
  File "/home/pi/pi-RFID/MFRC522-python/mfrc522/__init__.py", line 2, in <module>
    from .SimpleMFRC522 import SimpleMFRC522
  File "/home/pi/pi-RFID/MFRC522-python/mfrc522/SimpleMFRC522.py", line 3, in <module>
    import MFRC522
ImportError: No module named 'MFRC522'

death-droid, 2019-03-27:
Hi Matze,

I am unable to reproduce the issue you are having and am able to run the "read" sample without any issue using Python 3 and the latest version of the library.

`File "/home/pi/pi-RFID/MFRC522-python/mfrc522/SimpleMFRC522.py", line 3, in 
import MFRC522`

This indicates that you are using an old version of the library, that line was changed from

`import MFRC522`
to
`from . import MFRC522`

Which in turn fixed the importing issue with Python 3 while still retaining compatibility with Python 2.7.

closed 2019-04-18 by death-droid

Shreedhards7, 2023-08-17:
sudo python3 -m pip install mfrc522 --upgrade

Sure it will work try it !