pimylifeup/MFRC522-python #15: Continous reading of Tag gives None ever other time

MichaelMay81, 2019-06-26:
I want to detect when a tag is present and when it is removed. First I thought it might be a hardware issue, that my request frequency is to high. But even waiting for 10 seconds still gives me None every other time.

jaypeeig, 2019-07-10:
Same issue, I have to put the data on a queue list size of 3 to determine rfid is placed or not using the `read_no_block` function

asseeger, 2019-07-16:
I can confirm that this is an issue... In my app, I now just call read_id_no_block() twice to make it work:

```python
        while True:
            id = self.rfidController.reader.read_id_no_block()
            if id == None:
                id = self.rfidController.reader.read_id_no_block()
                if id == None:
                    logger.debug('no chip present')
            else:
               # do stuff (verify id etc.)
            time.sleep(1) 
```

mbagur, 2020-03-04:
I have the same issue and @asseeger 's trick is working for me too.