pimylifeup/MFRC522-python #16: Error reading uid[x] in HEX format

parovelb, 2019-08-02:
Dear Coders,
Congratulations for your work. I use your code to read tags with a RC522. Also I use a second MF7 reader for double checking the tags. 

**PROBLEM:** While testing the readings of UIDs of the tags, I found out there are missing "0" in some `uid[]` in the output of the RCC522 reader. The output of a single tag on both readers:
`pi@raspberrypi:~/Desktop/Biovalley $ python rfid_read_GIGATEK_MF7.py 
rfid_read.py -> rfid reader on port /dev/ttyUSB1
rfid_read.py -> tag hex AC8C5E0A
rfid_read.py -> tag dec  2894880266
pi@raspberrypi:~/Desktop/Biovalley $ python rfid_read_MFRC522.py 
Hold a tag near the reader. You have 5 s. Hurry up!
Card detected
bag tag UID: 10,94,140,172
bag tag endian: ('172', '140', '94', '10')
bag tag hex: ('0xac', '0x8c', '0x5e', '0xa')
bag tag hex string: ('ac', '8c', '5e', 'a')
bag tag hex string concatenated: ac8c5ea
bag tag dec: 180930026
`
As you can see the tag number in HEX of the MF7 is "AC8C5E0A". While the tag number in hex of the RC522 is "ac8c5ea". There is a "0" missing in the HEX transformation of "10" into "0xa". I have troubles finding the source of the problem in the code of the MFRC522.py and would really appreciate some help.

wilbolinux, 2019-08-07:
Hi great work but,

Have you found a solution for this Problem.. ? 
I have the same Problem... i have tags with UID hex "0" inside.. 
if i convert it from decimal to hexadecimal and
 the zero are not in the ID. :-/ 

Normally:
I take the long decimal number that i get from the example code read tag:

DecID  : 85433105273
--> change it to 
HEX     : 13e435bb79
---> and cut the last 2 chars...
**13e435bb** is the right UID.

but if i use this Tag:
i have a problem...

DecID  : 56302315998
HEX    : d1be129de
and cut the last 2 chars...
**d1be129** is a false UID.

The right UID was:
**0d1be129**

  ..How i doing wrong ?
....Can i use an other python script to solve this Issue ?

Thank you for your help,
Greetings Wilbo / Germany.  :-)

wilbolinux, 2019-08-07:
...hmmm was a possibility, I just thought of....

checkit if the UID has only 7 digits...  
and add as first digit a zero 
to set it up to 8 digits 

hmhm...

parovelb, 2019-08-08:
> ...hmmm was a possibility, I just thought of....
> 
> checkit if the UID has only 7 digits...
> and add as first digit a zero
> to set it up to 8 digits
> 
> hmhm...

Hi wibolinux!
I think adding a 0 will not solve the problem. As I understand the script reads 4 UIDs in blocks and the missing 0 could vanish in any of these 4. Perhaps I am wrong but if you check for 7 digits it is hard to guess where is the missing 0 is.

wilbolinux, 2019-08-08:
Youre right... of course ... unfortunately this code works with the rfid 522 perfectly without lockups... , but it has false reaults via the UID... :-/ at this point... useless for many situations..

parovelb, 2019-08-08:
Thank to the proposed [solution to the removed zeros on Stackoverflow](https://stackoverflow.com/questions/15884677/python-printing-hex-removes-first-0) I was able to solve the issue. Below is the working script with the use of `zfill` and `map`:
`
def Read_Bag_Tag():
    # read with MFRC522 on GPIO
    # define variables
    global value_rfid_2, reply_rfid_2, bag_tag, refTime
    
    # assign values
    refTime = datetime.now()
    
    # Create an object of the class MFRC522
    MIFAREReader = MFRC522.MFRC522()
    
    # read port
    while int((datetime.now()-refTime).seconds) < 5:
        # Scan for cards    
        (status,TagType) = MIFAREReader.MFRC522_Request(MIFAREReader.PICC_REQIDL)
    
        # If a card is found
        if status == MIFAREReader.MI_OK:
            print "Card detected"
        
        # Get the UID of the card
        (status,uid) = MIFAREReader.MFRC522_Anticoll()
    
        # If we have the UID, continue
        if status == MIFAREReader.MI_OK:
            # general rules
            print "bag tag UID: %s,%s,%s,%s" % (uid[0], uid[1], uid[2], uid[3])
            bag_tag_endian = (uid[3], uid[2], uid[1], uid[0])
            print 'bag tag endian:', bag_tag_endian
            bag_tag_hex = hex(uid[3]), hex(uid[2]), hex(uid[1]), hex(uid[0])
            print 'bag tag hex:', bag_tag_hex

            # using map
            bag_tag_map = map("{:02x}".format, (uid[3], uid[2], uid[1], uid[0]))
            print "bag tag map:", bag_tag_map
            bag_tag_map_hex = bag_tag_map[0], bag_tag_map[1], bag_tag_map[2], bag_tag_map[3]
            print "bag tag map hex:", bag_tag_map_hex
            bag_tag_map_hex_con = bag_tag_map[0] + bag_tag_map[1] + bag_tag_map[2] + bag_tag_map[3]
            print "bag tag map hex concatenated:", bag_tag_map_hex_con
            bag_tag_map_hex_con_dec = int(bag_tag_map_hex_con, 16)
            print "bag tag map hex concatenated dec:",bag_tag_map_hex_con_dec

            # using zfill
            bag_tag_hex2chr = hex(uid[3])[2:], hex(uid[2])[2:], hex(uid[1])[2:], hex(uid[0])[2:]
            print 'bag tag hex two char:', bag_tag_hex2chr
            bag_tag_zfill = hex(uid[3])[2:].zfill(2), hex(uid[2])[2:].zfill(2), hex(uid[1])[2:].zfill(2), hex(uid[0])[2:].zfill(2)
            print "bag tag hex two char zero fill:", bag_tag_zfill
            bag_tag_str = str(hex(uid[3])[2:].zfill(2))+str(hex(uid[2])[2:].zfill(2))+str(hex(uid[1])[2:].zfill(2))+str(hex(uid[0])[2:].zfill(2))
            print "bag tag hex string concatenated:", bag_tag_str
            bag_tag = int(bag_tag_str, 16)
            print 'bag tag dec:', bag_tag
            
            # This is the default key for authentication
            key = [0xFF,0xFF,0xFF,0xFF,0xFF,0xFF]
            
            # Select the scanned tag
            MIFAREReader.MFRC522_SelectTag(uid)
    
            # Authenticate
            status = MIFAREReader.MFRC522_Auth(MIFAREReader.PICC_AUTHENT1A, 8, key, uid)
    
            # Check if authenticated
            if status == MIFAREReader.MI_OK:
                MIFAREReader.MFRC522_Read(8)
                MIFAREReader.MFRC522_StopCrypto1()
            else:
                print "Authentication error"
            
            break
            
    GPIO.cleanup()      # Clear input buffer
`

wilbolinux, 2019-08-08:
Hi! I wrote a little patch to use
 the **SimpleMFRC522.py** :

(its located in : 
/usr/local/lib/python3.7/dist-packages/mfrc522/SimpleMFRC522.py

Replace the Definition Block :
**dev uid_to_num(self, uid)**

with this code:

  def uid_to_num(self, uid):
      n = 0
      for i in range(0, 5):
          n = n * 256 + uid[i]
      ## This Code translate the Decimal to hex and filled up with the vanish Zero 
      print((hex(uid[0])[2:].zfill(2)),(hex(uid[1])[2:].zfill(2)),(hex(uid[2])[2:].zfill(2)),(hex(uid[3])[2:].zfill(2)))
      return n

If i use the readtag python script, 
i have now following Output:

0d 1b e1 29
56302315998

(hexadecimal and decimal output of the UID)

Greetings
Wilbo.

wilbolinux, 2019-08-08:
it's not pretty but it runs.

closed 2019-08-12 by parovelb