pimylifeup/MFRC522-python #17: I don't Know how to put this?

PizzaRules668, 2019-08-26:
[image]

PizzaRules668, 2019-09-08:
how do I fix this?

tomatow, 2019-10-25:
Did you turn on SPI on RPi?

PizzaRules668, 2019-10-28:
yes

PizzaRules668, 2019-11-10:
What is the Command to make sure I have SPI is on?

ccoenen, 2020-07-03:
`lsmod | grep spi` should give you at least two lines of output if the SPI module is loaded. If it is not, you can enable it in the raspi-config menu.

PizzaRules668, 2020-07-04:
I looked in the dev folder to see If the was a spi file or folder, there was not a spi file or folder. So I reinstalled Raspbian and then there was a spidev0.0 and spidev0.1 files. Now when I run write.py(from pimylifeup's tutorial) it does not get past Now place your tag to write. it just stops, does not error out it just does not print Written.