pimylifeup/MFRC522-python #31: AUTH ERROR

MrTobyUp, 2020-11-09:
When I'm trying to read, I am getting an error saying:
"AUTH ERROR!!
AUTH ERROR(status2reg & 0x08) != 0"
Afterwards the UID is printed out correctly, but the Text is empty.
If I write onto a completly new card, and read afterwards, it works fine, but if I'm write onto the card with an mobile app, it's not possible to read.
Is there some kind of Authentication going on and can I somehow bypass it?

n1ywb, 2020-11-19:
it's probably because you're using a newer RFID tag. I'm having the same issue w NTAG215 tags. read somewhere they don't support auth, which would explain the error. but it's not reading the SN correctly either. I might try to hack in a fix.

n1ywb, 2020-11-19:
If you are using an NTAG and you just need the serial number, this hack will dump out the bytes that have it

```
#!/usr/bin/env python

import RPi.GPIO as GPIO
from mfrc522 import MFRC522
from mfrc522 import SimpleMFRC522
from time import sleep
from datetime import datetime

reader = MFRC522()

try:
    while True:
        status, _ = reader.MFRC522_Request(reader.PICC_REQIDL)
        if status != reader.MI_OK:
            sleep(0.1)
            continue
        status, backData = reader.MFRC522_Anticoll()
        buf = reader.MFRC522_Read(0)
        reader.MFRC522_Request(reader.PICC_HALT)
        if buf:
            print(datetime.now().isoformat(), ':'.join([hex(x) for x in buf]))
finally:
        GPIO.cleanup()
```

it doesn't trigger authentication so you don't get that error.

https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf has the secret decoder ring for turning those bytes into a serial number.

KevinCeyland, 2022-11-21:
> If you are using an NTAG and you just need the serial number, this hack will dump out the bytes that have it
> 
> ```
> #!/usr/bin/env python
> 
> import RPi.GPIO as GPIO
> from mfrc522 import MFRC522
> from mfrc522 import SimpleMFRC522
> from time import sleep
> from datetime import datetime
> 
> reader = MFRC522()
> 
> try:
>     while True:
>         status, _ = reader.MFRC522_Request(reader.PICC_REQIDL)
>         if status != reader.MI_OK:
>             sleep(0.1)
>             continue
>         status, backData = reader.MFRC522_Anticoll()
>         buf = reader.MFRC522_Read(0)
>         reader.MFRC522_Request(reader.PICC_HALT)
>         if buf:
>             print(datetime.now().isoformat(), ':'.join([hex(x) for x in buf]))
> finally:
>         GPIO.cleanup()
> ```
> 
> it doesn't trigger authentication so you don't get that error.
> 
> https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf has the secret decoder ring for turning those bytes into a serial number.
> 
> anticoll doesn't correctly return the serial number for ntags. it also doesn't finish the anticollision sequence, but oh well. basic reads work.

How can we get the badge id from your method?

MrTobyUp, 2022-11-25:
Little late to the party, but thx :)

closed 2022-11-25 by MrTobyUp

tslater, 2022-12-16:
Has anyone figured out how to actually read the data on a NTAG215?

scaevitas, 2023-09-17:
> If you are using an NTAG and you just need the serial number, this hack will dump out the bytes that have it
>
> ```
> #!/usr/bin/env python
>
> import RPi.GPIO as GPIO
> from mfrc522 import MFRC522
> from [...]

Hello. This is years later (so sorry) but when I run this code, it keeps saying error reading

is there a reason this happens and is there any way to fix it?

GentNisjes, 2024-03-16:
well i guess we'll have to accept the occasional AUTH ERROR ...

Lippsson, 2024-03-27:
Hello! 
Could you me help in my case. Sorry i´m not a developer and tried with chatGPT if i can modify the code that the auth error ignores from this code without success: 

`  def _read_card_legacy(self) -> str:
        error, tag_type = self.device.request()
        if not error:
            # Perform anti-collision detection to find card uid
            error, uid = self.device.anticoll()
            if not error:
                card_id = ''.join((str(x) for x in uid))
                if self.log_all_cards is True:
                    self._logger.debug(f"Card detected with ID = '{card_id}'")
                return card_id
        if self.log_all_cards is True:
            self._logger.debug("Error during reading card.")
        return ''

    def _read_card_normal(self) -> str:
        # The Cards UID can be stored as"
        #  - the actual integer UID number (decoded - the right way)
        #  - concatenated set of raw bytes (not decoded - this was previously the only way)
        # Only in legacy mode use raw bytes
        #
        # For the normal mode use the new convenience function
        #
        uid = self.device.read_id(as_number=True)
        # Check for empty string or NoneType
        if not uid:
            if self.log_all_cards is True:
                self._logger.debug("Error during reading card.")
            return ''
        card_id = str(uid)
        if self.log_all_cards is True:
            self._logger.debug(f"Card detected with ID = '{card_id}'")
        return card_id

    def read_card(self) -> str:
        # Scan for cards
        self.device.wait_for_tag()
        if not self._keep_running:
            return ''
        return self._read_function()`
        
        the code is from a media player project AlvinSchiller mentioned above. Thank you for you help.

y-nk, 2024-04-14:
@tslater i'm a noob in python but hopefully chatgpt is not.

i have some luck decoding the code with this snippet:

```py
import mfrc522

# Create an object of the MFRC522 class
mfrc522 = mfrc522.MFRC522()

# Main function
def main():
    try:
        while True:
            # Scan for tags
            (status, TagType) = mfrc522.MFRC522_Request(mfrc522.PICC_REQIDL)

            # If a tag is found
            if status == mfrc522.MI_OK:
                print("Tag detected")

                # Get the UID of the tag
                (status, uid) = mfrc522.MFRC522_Anticoll()

                # If the UID is successfully obtained
                if status == mfrc522.MI_OK:
                    print("UID: " + ":".join([str(x) for x in uid]))

                    # Select the tag
                    mfrc522.MFRC522_SelectTag(uid)

                    # Read data from the tag
                    data = [elem for index in [6] for elem in mfrc522.MFRC522_Read(index)]
                    result = ''.join([chr(charcode) for charcode in data])
                    print("Data read:", result)
                else:
                    print("Error obtaining UID")

    except KeyboardInterrupt:
        print("Exiting...")
        mfrc522.MFRC522_StopCrypto1()

if __name__ == "__main__":
    main()

```

you'll notice the `elem for index in [6] for elem` as in "it's weird to have [6]` - i'm still trying to make sense of it and currently manually modify the index list to explore the tags i had set with my android phone - i'm saying this because of https://stackoverflow.com/questions/59515271/why-android-nfc-reader-adds-en-before-the-message

cheers

UtkarshRana22, 2024-12-11:
How to actually write data to the card without getting the auth error I have literally tried everything 

Please help

y-nk, 2024-12-11:
@UtkarshRana22 u tried gpt?

UtkarshRana22, 2025-01-04:
> @UtkarshRana22 u tried gpt?

I did but I got a better by myself I can post it here if you want though it just helps in reading UID associated with RFID card without throwing the auth error. I think the code above does that successfully too

Lippsson, 2025-01-04:
> I did but I got a better by myself I can post it here if you want though it just helps in reading UID associated with RFID card without throwing the auth error. I think the code above does that successfully too

Also intressted to see your solution