pimylifeup/MFRC522-python #32: Reader not reading?

BlockTinkerer187, 2020-12-28:
Hello.

I have a RC522 module in hopes of having it read cards, but to no avail will it read the provided tags. I have the following code:
```python
#!/usr/bin/env python

import RPi.GPIO as GPIO
import mfrc522
from mfrc522 import SimpleMFRC522

reader = SimpleMFRC522()

try:
        print("Hold a tag near the reader")
        id, text = reader.read()
        print("ID: %s\nText: %s" % (id,text))
        print()
finally:
        GPIO.cleanup()
```

When run with `python3 ./read.py` it reads `Hold a tag near the reader`. Yet, despite my endless waving next to the reader, it does not detect the card. Is there anything I can do to make sure it is connected properly and troubleshoot? Could it possible I have a defective module?

Thanks!

solderdot72, 2020-12-29:
Using a Raspberry Pi 3B+ this code works fine under Python 3 (Python 3.7.3 (default, Jul 25 2020, 13:03:44))
However, under Python 2.7 (Python 2.7.16 (default, Oct 10 2019, 22:02:15)) I observe the very same issue: The read method does not terminate if a RFID tag is presented to the reader.

So for some reason this does not work for Python 2.7.

I would greatly appreciate hints on how to make this work with Python 2.7.

BlockTinkerer187, 2020-12-31:
> Using a Raspberry Pi 3B+ this code works fine under Python 3 (Python 3.7.3 (default, Jul 25 2020, 13:03:44))
> However, under Python 2.7 (Python 2.7.16 (default, Oct 10 2019, 22:02:15)) I observe the [...]

I ended up switching to my 3B+ only to find I am still encountering the issue. I am using the latest version of Raspberry Pi OS with the Desktop Environment.

May I have some information about the Pi and the OS it is running? Also some information about your reader and the whereabouts of it's purchase would be amazing. I am trying to deduce if it is a software issue or a hardware issue.

solderdot72, 2020-12-31:
I am running my software under the December 2020 Raspbian Buster distribution.  

```
$ uname -a
Linux $$$ 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
```

Don't want to do advertisement, but here's the product I purchased on Amazon (German vendor):

AZDelivery 5 x RFID Kit RC522 mit Reader, Chip und Card 13,56MHz I2C IIC SPI kompatibel mit Arduino und Raspberry Pi inklusive E-Book!

For a quick test I did this sequence:
```
python3
>>> from mfrc522 import SimpleMFRC522
>>> reader = SimpleMFRC522()
>>> reader.read()

```

ncls-alien, 2024-07-21:
Having the same issue using a Pi 5 with Python 3.11.2 installed. It works fine until the `reader.read()` and then just doesn't react to the tag I put near the reader. Same goes for writing the tag.