pimylifeup/MFRC522-python #51: Impossible to install the Library on Thonny

Nicolebendo, 2022-05-12:
Hello everyone, I'm pretty new to this and have close to no knwoledge in Installing Libraries.
When I try to install this library on Thonny, it says that it requires Visual Studio (that I installed, as well as Visual Studio Build Tools 2022).
Now it gets me this error: 
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.32.31326\\bin\\HostX86\\x86\\cl.exe' failed with exit status 2

I've searched on the internet for a solution but with no success...
Thank you

nmcc1212, 2023-01-12:
Are you using windows?

Anonymouse83, 2024-11-29:
I have a similar issues installing libraries - including pprint, and I am running Ubuntu