pimylifeup/MFRC522-python #61: Issue with Raspberry Pi 5

chetan1836, 2023-12-27:
When I Tried to run SimpleMFRC522() function I get the following error on Pi 5.
`raceback (most recent call last):`
`  File "/home/pi/Documents/rfid_test/efid_read.py", line 4, in <module>`
`    reader = SimpleMFRC522()`
`  File "/home/pi/.local/lib/python3.7/site-packages/mfrc522/SimpleMFRC522.py", line 14, in __init__`
`    self.READER = MFRC522()`
`  File "/home/pi/.local/lib/python3.7/site-packages/mfrc522/MFRC522.py", line 151, in __init__`
`    GPIO.setup(pin_rst, GPIO.OUT)`
`RuntimeError: Cannot determine SOC peripheral base address`

tinotino83, 2024-07-11:
I have the same problem.
Can this be fixed?

ncls-alien, 2024-07-21:
Uninstall `RPi.GPIO` and install `rpi-lgpio` instead.